Ytelse/pcb #15: Maybe use an FT232 dongle instead of RS232

kringle, 2016-09-27:


kringle, 2016-09-27:
UART-USB chip: http://www.ftdichip.com/Support/Documents/DataSheets/ICs/DS_FT232R.pdf, even available on Vault. Can be viable alternative to USB-RS232

seel93, 2016-09-28:
Dette kan eventuelt avklares med gunnar på neste møte. Kan se på hvordan den implementeres slik at det går fort å endre, dersom denne komponenten er mer optimal.

kringle, 2016-09-28:
Joa, men det som er greia er at dersom me bruker RS232 må me ha et adapter fra rs232 til USB, altså typisk en FT232. Så, enten blir linken: UART->RS232->eksternRS232tilUSB->USB, eller så blir den UART->USB. FT232 er heller ikkje vanskelig eller tidkrevende å koble opp, så det er minst like enkelt samt at man får det direkte på vanlig USB med en velprøvd driver. Mange fordeler...

seel93, 2016-09-30:
CBUS3: Configurable CBUS I/O Pin. Function of this pin is configured in the
device internal EEPROM. Factory default configuration is PWREN#. See
CBUS Signal Options, Table 3.99. PWREN# should be used with a 10kΩ
resistor pull up.

seel93, 2016-09-30:
check if pull down resistor needs 5V or 3V3, and if the UART can run on 3V3